Panda381/PicoVGA #1: Does it work with reference design wiring or custom wiring?

dglaude, 2021-06-18:
Thank you for this code and the great YT video demonstrating it.
I was wondering if those were supposed to work with Raspberry reference design for attaching a VGA to a PICO.
Mine is the implementation of Pimoroni (that is supposed to respect that mapping): https://shop.pimoroni.com/products/pimoroni-pico-vga-demo-base

But right now it fail on me (could by my screen or Pico or Pico demo).

Thanks.

PS: If it's an alternate design maybe you would directly spot what need to be alter for all those example to work, it is mostly a matter of picking the right GPIO for the same usage.

dglaude, 2021-06-18:
FYI:
Here are the demo for the reference design: https://github.com/raspberrypi/pico-playground
Here is the reference design: https://datasheets.raspberrypi.org/rp2040/hardware-design-with-rp2040.pdf

Panda381, 2021-06-18:
Hi, unfortunately PicoVGA cannot be compatible with Pimoroni hardware. Used 
PIO controller requires a continuous section of GPIO pins. The pins can only
be moved as a whole section, but their order cannot be changed. 16-bit and 8
-bit VGA require differently wired hardware and are not interchangeable. 
They cannot be interchanged by changing the software.

---------- Původní e-mail ----------

Komu: Panda381/PicoVGA ***@***.***>
Datum: 18. 6. 2021 15:13:28
custom wiring? (#1) 
"

pico-playground(https://github.com/raspberrypi/pico-playground)
hardware-design-with-rp2040.pdf
(https://datasheets.raspberrypi.org/rp2040/hardware-design-with-rp2040.pdf)

(https://github.com/Panda381/PicoVGA/issues/1#issuecomment-864029455), or 
unsubscribe
"

dglaude, 2021-06-19:
Thank you, I understand (now).
I will have to do a bit more electronic than expected to enjoy this.
Any chance of a gerber file or some "ready to build" PCB. :-)

This seems very interesting that you managed to get a lot of the work done by the PIO processor, even if with very few colours.

Panda381, 2021-06-19:
In simple way should be sufficient connect a few resistors to Pico pins and 
VGA connector: http://www.breatharian.eu/hw/picovga/img/PicoVGA_diagram2.jpg
(http://www.breatharian.eu/hw/picovga/img/PicoVGA_diagram2.jpg)

I am working on retro computer using Raspberry Pico and PicoVGA library. It 
will enable running Pico programs from SD card and (later) run some 
emulators of retro computers. I think base electronics will be ready for 
about a month to publish.

Nemecek

---------- Původní e-mail ----------

Komu: Panda381/PicoVGA ***@***.***>
Datum: 19. 6. 2021 17:15:57
custom wiring? (#1) 
"