Panda381/PicoVGA #3: ScreenSegmAText crashes PicoVGA

codaris, 2021-12-14:
I've been trying to use `ScreenSegmAText` but it appears to crash PicoVGA.

I've tried with a few simple programs without luck.

As as test, I modified the matrix rain sample using `ScreenSegmAText()` and it also doesn't work:

    ScreenSegmAText(g, TextBuf, Font_Copy, FONTH, DefPal16, TEXTWB);

Expected output would be the matrix rain but with random CGA colors instead I get no monitor sync at all.

Panda381, 2021-12-14:
Rendering ScreenSegmAText is more time consuming than ScreenSegmFText and cannot render full 640px resolution on VGA (on TV it could be OK). You must use lower segment width to render it.

codaris, 2021-12-14:
That would explain it.  Thanks for the quick reply.  My target resolution is 640x480.  It looks like ScreenSegmCText() might be a suitable alternative for my needs.

closed 2021-12-14 by codaris