Panda381/PicoVGA #7: How to build on non Windows ?

jonshouse1, 2022-04-29:
This looks great as a project.

The build system seems very Windows specific, will this build with the Linux pico gcc tools ?
I have a "VGA demo base" for the Pico, how does the pinout compare, how do I configure it ?

Thanks.

Panda381, 2022-04-29:
Maybe someone can modify it for Linux, I can't. The VGA demo board cannot be used for 8-bit output, PIO requires a continuous pin sequence.

jonshouse1, 2022-04-29:
>Maybe someone can modify it for Linux, I can't.

Fair enough. I would like to give it a go, but at the moment I a beginner with the pico and struggling with the tools myself.  I've got a linux virtual machine setup with the toolchain working. I can build the VGA demos for this board and have tested them.
It would be much more interesting to run your games, they look great, but I don't really want multiple build systems if I can avoid it so I would prefer not to have to install windows just to build it. 

>The VGA demo board cannot be used for 8-bit output, PIO requires a continuous pin sequence.

Ok, not sure I fully understand. I can confirm this code does work on the VGA demo board.
https://github.com/raspberrypi/pico-playground

 Are you saying that some aspect of the PIO code on your project requires a different pinout ?

The VGA demo board is figure 24 (Page 22) from this document
https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf

Panda381, 2022-04-29:
Apparently in Linux there is a problem with referring to the "current directory" as it comes from the "home directory". The output color components cannot be remapped to different pins, because all 8 outputs must form a continuous sequence (this is a PIO limitation).

jonshouse1, 2022-04-29:
I've made some progress. Managed to get vga.pio.h built.  The pioasm utility is a bit crap as the Linux version does not tolerate the dos style CRLF line terminators.
I did the following

```
Built pioasm
        cd /pico/pico-sdk/tools/pioasm
        cmake .
        make
        sudo cp pioasm /usr/bin

pioasm sulks about DOS style newline
        apt install dos2unix
        cd PicoVGA/_picovga
        dos2unix vga.pio

        cd PicoVGA/_picovga
        pioasm ../_picovga/vga.pio vga.pio.h
```

jonshouse1, 2022-04-29:
>If you remap the top 3 bits to red2..4, the red0 red1 will have top 2 bits of the green component.

Ah ok.  If this cant be fixed in software then all I can do is to make a bodge board to sit between the Pico module and the VGA base with patch wires, nasty though, I would rather modify the code if possible as a couple of hundred of these demo boards have already been sold and other people would probably also like to play your games.

jonshouse1, 2022-04-29:
Getting closer to building.   I've found elf2uf2 source in several places, I will investigate some more later.

```C++ ../_picovga/vga_pal.cpp
C++ ../_picovga/vga_screen.cpp
C++ ../_picovga/vga_util.cpp
C++ ../_picovga/vga_vmode.cpp
C++ ../_picovga/util/canvas.cpp
C++ ../_picovga/util/mat2d.cpp
C++ ../_picovga/util/overclock.cpp
C++ ../_picovga/util/print.cpp
C++ ../_picovga/util/rand.cpp
C++ ../_picovga/util/pwmsnd.cpp
C++ ../_picovga/font/font_bold_8x8.cpp
C++ ../_picovga/font/font_bold_8x14.cpp
C++ ../_picovga/font/font_bold_8x16.cpp
C++ ../_picovga/font/font_boldB_8x14.cpp
C++ ../_picovga/font/font_boldB_8x16.cpp
C++ ../_picovga/font/font_game_8x8.cpp
C++ ../_picovga/font/font_ibm_8x8.cpp
C++ ../_picovga/font/font_ibm_8x14.cpp
C++ ../_picovga/font/font_ibm_8x16.cpp
C++ ../_picovga/font/font_ibmtiny_8x8.cpp
C++ ../_picovga/font/font_italic_8x8.cpp
C++ ../_picovga/font/font_thin_8x8.cpp
C++ ../_sdk/new_delete.cpp
ld build/program.elf
bin program.bin
hex program.hex
lst program.lst
sym program.sym
siz program.siz
uf2 program.uf2
make: execvp: ../_exe/elf2uf2.exe: Permission denied
make: *** [../Makefile.inc:435: program.uf2] Error 127
```

jonshouse1, 2022-04-29:
Ok, got it built.

First I needed to build the sdk version of elf2uf2

```
cd pico/pico-sdk/tools/elf2uf2
cmake .
make
```

Then in your Makefile.inc I changed:

```
# uf2
#UF = ../_exe/elf2uf2.exe
UF = $$PICO_SDK_PATH/tools/elf2uf2/elf2uf2
```

Now I can build pacman

```CC ../_tinyusb/portable/raspberrypi/rp2040/hcd_rp2040.c
CC ../_tinyusb/portable/raspberrypi/rp2040/rp2040_usb.c
CC ../_tinyusb/tusb.c
C++ src/data.cpp
C++ src/game.cpp
C++ src/main.cpp
C++ src/open.cpp
C++ img/logo.cpp
C++ img/tiles.cpp
C++ img/sprites.cpp
C++ img/status.cpp
C++ snd/startsnd.cpp
C++ snd/siren.cpp
C++ snd/wakka.cpp
C++ snd/frighten.cpp
C++ snd/eatghost.cpp
C++ snd/eatfruit.cpp
C++ snd/death.cpp
C++ snd/extrapac.cpp
C++ snd/intermission.cpp
C++ ../_picovga/vga.cpp
C++ ../_picovga/vga_layer.cpp
C++ ../_picovga/vga_pal.cpp
C++ ../_picovga/vga_screen.cpp
C++ ../_picovga/vga_util.cpp
C++ ../_picovga/vga_vmode.cpp
C++ ../_picovga/util/canvas.cpp
C++ ../_picovga/util/mat2d.cpp
C++ ../_picovga/util/overclock.cpp
C++ ../_picovga/util/print.cpp
C++ ../_picovga/util/rand.cpp
C++ ../_picovga/util/pwmsnd.cpp
C++ ../_picovga/font/font_bold_8x8.cpp
C++ ../_picovga/font/font_bold_8x14.cpp
C++ ../_picovga/font/font_bold_8x16.cpp
C++ ../_picovga/font/font_boldB_8x14.cpp
C++ ../_picovga/font/font_boldB_8x16.cpp
C++ ../_picovga/font/font_game_8x8.cpp
C++ ../_picovga/font/font_ibm_8x8.cpp
C++ ../_picovga/font/font_ibm_8x14.cpp
C++ ../_picovga/font/font_ibm_8x16.cpp
C++ ../_picovga/font/font_ibmtiny_8x8.cpp
C++ ../_picovga/font/font_italic_8x8.cpp
C++ ../_picovga/font/font_thin_8x8.cpp
C++ ../_sdk/new_delete.cpp
ld build/program.elf
bin program.bin
hex program.hex
lst program.lst
sym program.sym
siz program.siz
uf2 program.uf2

$ ls -l program*
-rwxrwxr-x 1 jon jon  452760 Apr 29 18:28 program.bin
-rw-rw-r-- 1 jon jon 1273560 Apr 29 18:28 program.hex
-rw-rw-r-- 1 jon jon 1567404 Apr 29 18:28 program.lst
-rw-rw-r-- 1 jon jon 1576749 Apr 29 18:28 program.map
-rw-rw-r-- 1 jon jon     107 Apr 29 18:28 program.siz
-rw-rw-r-- 1 jon jon   23153 Apr 29 18:28 program.sym
-rw-rw-r-- 1 jon jon  905728 Apr 29 18:28 program.uf2
```

It is going to take some more work to make the Linux build smoother but the changes are minor.

jonshouse1, 2022-04-29:
I've tried the executable.  The UART is active and prompts me so the game is running.  The monitor is black but thinks it has some sort of sync.  I will have a go at hacking the PIO code around, not sure I understand anything much at the moment.  I will read it, where does "pins" come from, at some point something must map the bytes or bits to the GPIO pins - where should I look?

I'f I can make the changes minor and write a README.linux file can you import those changes back into your project or would you rather I fork it?

Panda381, 2022-04-29:
PicoVGA (in the basic unmodified version) uses only the combined CSYNC (HSYNC) synchronization, which connects to the monitor's HSYNC signal. The VSYNC monitor signal must remain unconnected. Most monitors support this mode, but some older monitors do not. Furthermore, PicoVGA assumes that the HSYNC signal follows immediately after the color signals on the GPIO. It is possible to remap it to another GPIO, but it is not ready without code modifications. You have to define VGA_GPIO_SYNC in vga_config.h and then do some more code correction in the VgaPioInit() function in the vga.cpp file.

jonshouse1, 2022-04-29:
> PicoVGA (in the basic unmodified version) uses only the combined CSYNC (HSYNC) synchronization, which connects to the monitor's HSYNC signal. The VSYNC monitor signal must remain unconnected. Most monitors support this mode, but some older monitors do not. Furthermore, 

Interesting you should mention that, I was reading the description on your web page for the project (nice writreup, thanks) and scratching my head.  I've done some dirty things with video in the past, but nothing in the specs say that VGA supports composite sync at all. I don't quite follow the schematic and description, not sure what the connection between audio and V-SYNC is about - it isn't labelled as such, does that mean no sound in VGA mode or that audio has a 60Hz hum and video breaks up if the audio data gets towards DC levels, i'm very confused by it.  You did a good demo video so clearly it works.
Maybe it the web page needs re-writing for that section, with separate descriptions of the possible video output modes and what is going on in each mode.     

Some suggestions:
"Computer monitors support CSYNC mixed sync"
maybe should be
"Some VGA monitors all support composite sync on the H-SYNC pin"
I feel you may just have been lucky with the monitors you tested.

I have a lot of pico specific reading to do before I can make a stab at any constructive changes, thanks for publishing the project, looks great. Do you have your own hardware available?

>PicoVGA assumes that the HSYNC signal follows immediately after the color signals on the GPIO.

The GPIO DAC is steady state, not sure what you are telling me here, the state of the analog video lines does not matter during line or frame sync but is typically black, I don't think sync on green was ever a requirement for VGA.

Panda381, 2022-04-30:
The original design used the VSYNC pin of the VGA connector as the audio output. This allowed a SCART connector to be connected to the TV without the need for a special connector. Plugging in the audio jack disconnected the audio output to the VGA connector. I used this design mainly with the ATpad game console. http://www.breatharian.eu/hw/atpad/index.html All of the 8 or so monitors I tried supported CSYNC composite sync. With the PicoVGA project, some users have occasionally reported that their monitor does not support composite CSYNC. In these cases, the PicoVGA library needs to be modified to support separate sync signals - the modification is described here: https://forums.raspberrypi.com/viewtopic.php?t=313634&p=1878457#p1878457

jonshouse1, 2022-04-30:
>All of the 8 or so monitors I tried supported CSYNC composite sync, I found this specification in the description of the VGA connector signals.

Do you have evidence for this, I've never seen even a tiny sniff of evidence that VGA devices are expected to support composite sync?

I tried to trace out the front end from the simplest VGA CRT monitor I could find, I gave up in the end as it seems the sync section is quite complicated, but best I can tell it is a pair of XOR gates with the CMOS Hsync/VSync signal pulled up to +5 via 27k, coupled into one half of the XOR via 330R.  The arrangement between the HSync processor (MC1391) and the main defection logic and PLL (TDA1170N) is above my pay grade.  I suspect the fact that most monitors worked with composite sync was just pure luck that they all chose to copy the same front end design and not use the sync arrangement documented in the TDA application notes.

See schematic page 33
http://www.bitsavers.org/pdf/samsung/monitors/08146-400-040_ML467_ML267_VGA_Mono_Monitor_Service.pdf

Personally I would not ship something with VGA output that relied on the monitor supporting composite sync, to do so is to confuse luck with specification, also I suspect the older and more vintage the monitor the less likely it may be to work, but that is only a guess.

>ATpad game console. http://www.breatharian.eu/hw/atpad/index.html 

That is very very impressive.

>I am using PicoVGA on my mini-computer, but it is not finished yet. https://www.oldcomp.cz/download/file.php?id=12823

As is this, nice work :-)

You are doing great work, your software is fantastic, but we may just have to agree to differ on the wisdom of composite sync for VGA outputs !

When it comes to specification vs "what works" I found a very similar thing with VGA driving SCART.  I came up with a simple way to drive RGB SCART from a Raspberry Pi board using a VGA adapter, works perfectly but the sync arrangement is way outside of what should work, looks great on every TV I tried it on, but again due to luck of implementation rather than meeting the specification.

Panda381, 2022-05-01:
I only use current LCDs, not CRTs. Maybe that's the difference. With this project, I've encountered more backlash than not supporting older CRTs - rather, HDMI is already required, and current TVs no longer support SCART, so this project is on the wane and I didn't want to improve it further.

jonshouse1, 2022-05-01:
>I only use current LCDs, not CRTs. Maybe that's the difference. With this project, I've encountered more backlash than not supporting older CRTs - rather, HDMI is already required, and current TVs no longer support SCART, so this project is on the wane and I didn't want to improve it further.

Fair enough.  I don't think you get backlash, just a bit of confusion and a little technical disagreement at some of the choices you made, but much respect for the work, it is very impressive.
I think some of the confusion about VGA signals might be that the most common documents google links for VGA specification are often a variation on descriptions of how to drive workstation (RGB sync on green) monitors from VGA outputs. This was of great interest in the early days of VGA but not a great description of VGA itself.
All noted on HDMI, I've yet to put any effort into understanding that interface, one day maybe.

Thanks for the great work and the very useful and comprehensive web pages and demos you have produced, that will keep me reading and learning for many hours.
I look forward to seeing what you do with your retro machine with the keyboard.

Do you want me to fork this project and add Linux build files or do you want me to Make changes and a Linux specific README and feed those changes back to you ?

Panda381, 2022-05-01:
Unfortunately I don't know how to use GitHub (it's too "Linux approach" which I cannot handle), my maximum was to figure out how to push a project to GitHub, but I can't make changes to it. :-(

jonshouse1, 2022-05-01:
>Unfortunately I don't know how to use GitHub (it's too "Linux approach" which I cannot handle)

I was in much the same situation last year.  I am still a git/github novice.  I just try and remember to zip the project up before I try and use git in cause I get it wrong.
Fortunately fork is dead simple as it is just a button push on the github web page, not sure what I do after that but I will figure it out. I will do that and make the changes for Linux building and add another readme specifically for that, but leave the windows batch files etc in place.

15 people have already forked your project, but best I can tell none have modified it to build on Linux, or if they did they did not change the description/readme.  See Forks under "about" on the right hand side of your github project page.

codaris, 2023-03-12:
I have create a fork of PicoVGA that builds on Linux:

https://github.com/codaris/picovga-cmake

I created this fork so that I could use the latest Pico SDK and compile PicoVGA for the Pico W (for Wifi support).  In this fork, I've modified all the examples and updated the documentation accordingly.  If anyone tries this fork and has an issue, please leave a ticket in that project.