Panda381/PicoVGA #12: SPI?

lynch-mob, 2022-11-22:
Are you using both SPI buses for this project? Will this project work with the "no OS SD" SD card driver if not? [https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico) Thank you for your time.

Panda381, 2022-11-22:
PicoVGA does not use the SPI interface and should work with the SD project. If the VGA pins need to be moved to free up the SD pins, just edit the VGA_GPIO_FIRST parameter. However, the condition remains that the pins for VGA must form a continuous row, they cannot be split. PicoVGA requires 8 DMA channels, leaving 4 DMA channels free for the SD project.

lynch-mob, 2022-11-22:
Thank you so much! BTW, this is one of the most impressive projects surrounding the Pico I have ever seen! Keep up the great work!

Panda381, 2022-11-23:
Thank you very much. :-)