Panda381/PicoVGA #13: Work with this circuit?

carlni2008, 2022-12-11:
Hi I am trying to get a rca output to work with this circuit:

[image]

This circuit works by combining the five VGA signals (R, G, B, H and V) into a crude approximation of a monochrome [RS-170] signal. The resistors at the top do a rough weighting of the three color channels to approximate how the eye responds to intensity, and the two resistors at the bottom add a combined sync signal to the mix.

I have tried this circuit diagram with the vga output of the pico, so I can have 2 video out ports on my pico. However only the vga port worked, and the rca didn't. 
I am not sure about the problem, but the possible problem might be:
1. I used regular ground for BGND since the pico doesn't have a individual ground for each color.
2. The pico only have CSYNC
In the documentation, it also states "The circuit needs VGA to TV drivers which output negative polarity HSYNC and VSYNC signals"

Do you have any ideas how I can make this work?

Panda381, 2022-12-11:
If you want monochrome TV signal, only connect all 3 RGB outputs together (resistors are not needed) and connect CSYNC to it via resistor 1K. Maybe some resistor will be needed between output and ground to lower voltage if image is too bright, but it is not necessary. And use TV mode of PicoVGA to have correct timmings.

carlni2008, 2022-12-11:
Hi, how do I change PicoVGA to TV mode?

Panda381, 2022-12-11:
Initialize videomode with DEV_PAL or DEV_NTSC device:
Video(DEV_NTSC, RES_EGA, FORM_8BIT, Box);
Or when using init with VgaCfg function, select video timings with some TV mode: Cfg.video = &VideoNTSC;

carlni2008, 2022-12-11:
That seems a bit complicated. By changing the mode of PicoVGA to TV mode for monochrome TV sininal so it will have the correct timing, that also means the timing for the vga port will be wrong. Are there a way so that both of the video ports will be useable without changing the program? I preferr the hardware solution more than the software solution.

carlni2008, 2022-12-12:
One more problem, i don't know why, some displays doesn't work with the vga. It might be the problem on how I connected the wires.

Panda381, 2022-12-12:
VGA and TV use different video signal timings and are not interchangeable by hardware modification. If you want to use a program without changing the program, you must provide mode switching in software - in response to a configuration switch or buttons.

Some VGA displays (especially older ones) do not support CSYNC composite synchronization. This requires modifying PicoVGA to generate a separate sync (on 2 pins) instead of composite sync. The modification is not yet included here in the source code, but I have roughly described it here: https://forums.raspberrypi.com/viewtopic.php?t=313634&start=25#p1878457

carlni2008, 2022-12-12:
ok I see

carlni2008, 2022-12-21:
> If you want monochrome TV signal, only connect all 3 RGB outputs together (resistors are not needed) and connect CSYNC to it via resistor 1K. Maybe some resistor will be needed between output and ground to lower voltage if image is too bright, but it is not necessary. And use TV mode of PicoVGA to have correct timmings.
> 
> Common ground is OK. Capacitor is not needed.

I just realized, by 3 RGB output, you mean the all 8 pin out put from the rpi pico?

Panda381, 2022-12-21:
In my experiments, I just connected all 4 signals together (RED+GREEN+BLUE+CSYNC), fed them to the video input of the TV, and it worked. For a first attempt, that should be enough. The picture was a bit overblown, so if one wanted to do it better, the resistor values would have to be better recalculated. But probably just adding one more resistor between the video signal and ground (I'm guessing 100 ohms or something, I didn't calculate it) should be maybe fine.

carlni2008, 2022-12-21:
ok thanks

carlni2008, 2022-12-22:
I am planning on keeping both a VGA connecter and a RCA connector with a switch to switch between the 2 modes. The problems  is that the 4 signals are connected for monochrome signal to work, but since the they are connected, the vga wont work. How should I wire this properly? 

Current attempt:
[image]

Panda381, 2022-12-22:
I hope that your wiring diagrams do not correspond to the real thing, but are just misrepresented. :-) If you want both VGA and RCA output at the same time, you need to use 2 sets of resistors. 1 set will be RGB output, the other will connect together to RCA output. The one for RCA will have different resistor values, which will be calculated to match the voltage levels (0 to 1V in 75 ohm load, 0.25V black level). GPIO outputs must be switched to maximum current - there is no function in SDK for this, you have to use the CPU control port directly).

carlni2008, 2022-12-22:
So what is the correct way to draw this diagram? Can you please draw it out for me? I am really a noob.

Panda381, 2022-12-22:
This wiring should work. I recalculated the resistor values. 
[image]

carlni2008, 2022-12-22:
My pico:
[image]
VGA: 
[image]
RCA:
[image]
This should work according to your schematics. The only thing I am worried is the Reset switch and the VSYNC being on the same pin, will this work?

carlni2008, 2023-02-04:
Hi, I have my PCB finished. There are a few problems, such as Vsync is on gpio 26, but I'll just use csync only for now. Since my pinout is differet, when modifying your test programs, I'll need to change the vga settings. So in vga_config.h, I changed the vga port pins to this:
[image]
After I finished compiling the code with c.bat, the gerated uf2 doesn't seems to be working, there is nothing showing on the monitor when I connected the VGA cable. Are there any configs I missed?

Panda381, 2023-02-05:
Check the sound generator (in the pwmsnd.cpp file) - it is set to GPIO19 by default, maybe it conflicts with the vga output.

carlni2008, 2023-02-08:
ok it works now, thank you so much

carlni2008, 2023-02-08:
Ok 2 problems:
1: You said to set VSYNC pin to output, which I have no idea how in C++ since I used to program my pico with python.
2. Is it possible to set the VSYNC to another GPIO pin that is not in the same sequence as the other pins(RGB and CSync = GPIO 14-22, and VSync is GPIO 26)
Thank you very much and I appologize for beening such a noob.

Panda381, 2023-02-09:
To set VSYNC to output, add this to function VgaPioInit() :

	// set VSYNC output
	gpio_init(VGA_GPIO_VSYNC);
	gpio_put(VGA_GPIO_VSYNC, 0);
	gpio_set_dir(VGA_GPIO_VSYNC, GPIO_OUT);

and add this to function VgaLine, before "switch (linetype)":

	if (CurVmode.vsync) gpio_put(VGA_GPIO_VSYNC, (linetype == LINE_VSYNC) ? 0 : 1);
 
Set VGA_GPIO_VSYNC to any pin (need not be sequence):

#define VGA_GPIO_VSYNC	xxx

carlni2008, 2023-02-11:
Error:
##

C++ ../_picovga/vga.cpp
../_picovga/vga.cpp: In function 'void VgaLine()':
../_picovga/vga.cpp:402:15: error: 'struct sVmode' has no member named 'vsync'; did you mean 'hsync'?
  402 |  if (CurVmode.vsync) gpio_put(VGA_GPIO_VSYNC, (linetype == LINE_VSYNC) ? 0 : 1);
make: *** [../Makefile.inc:458: build/vga.o] Error 1
Press any key to continue . . .

Panda381, 2023-02-11:
Sorry, that was from my other project. Change - delete the condition and just use this:

gpio_put(VGA_GPIO_VSYNC, (linetype == LINE_VSYNC) ? 0 : 1);