Panda381/PicoVGA #14: Single unified example that containing as much sub-examples as possible!

arduino12, 2023-01-21:
**Hi,**

**Amazing work!** it all works out of the box!
[image]

**1.** I want to have a unified UF2 program demonstrating as much sub-examples as possible on a 16MB RP2040 (like **[this](https://www.aliexpress.com/item/1005004005660504.html)**).
That way switching between examples will be easy as sending character to the serial interface or by pressing the BOOT button.
Do you plan on making such program? can I help?

**2.** I see you wrote about [loading UF2 programs out of a SD card](https://github.com/Panda381/PicoVGA/issues/1#issuecomment-864459927) - sounds great!
Is it ready? Can you share your work?

**3.** For now I hold the BOOT button, while pressing and releasing the RESET button in order to upload new UF2 program,
Is it possible for the examples to keep running the USB Storage Device - so we can upload new UF2 without entering the boot mode?
Else, is it possible for the examples to enter the boot mode by sending '^' char on the virtual USB-Serial command?
I thought of connecting a USB-UART bridge, and use its flow-control lines to toggle the BOOT and RST lines-
(So I can change UF2 programs via a PC script) but it will require 2 USB ports so I think its better to solve it in the software..

**4.** After have a working hardware with the ability to switch examples in a button press / serial command,
I will design a small PCB for it and would love to share it here if possible :)

**Thanks again for this repo and docs!
Arad** 👍

Panda381, 2023-01-21:
I am still working on a Pico microcomputer, which allows running programs from an SD card (similar to classic 8-bit computers). I'm creating a new custom SDK library for Pico, with much expanded features and easier to use - such as floating math with high precision (possibility to create calculator with Pico with optional calculation precision up to 300 digits).

arduino12, 2023-01-21:
**Wow!
OK I see so I will wait patiently :)**

#### I will try combining those examples and open a PR if it works:
Balloons 
Earth 
Hypno
Level Meter
Lines 
Matrix Rain
Oscilloscope 
Pi 
Spheres 
Spots 
Twister 
Water Surface

Panda381, 2023-01-21:
It may be difficult to combine programs because they use different settings during compilation time. At a minimum, you will need to share a frame video buffer.