Panda381/PicoVGA #15: Best way to update the PICO SDK?

codaris, 2023-01-31:
Hello Miroslav,

I'm using this project as a graphics processor for an 8bit computer and it works great (I use the remaining pins and remaining PIO to connect to a 6502 8bit computer bus).  I'm also using the Pico for USB keyboard and mouse support in the same project.  In this work, I noticed that the Pico SDK included in this library is out of date.  For my USB support, I manually updated the TinyUSB library to the latest version and just patched out anywhere there was a conflict with the older Pico SDK and that has worked.

However, I recently purchased a Pi Pico W for it's Wifi support and I'd like to use it with this project (giving my 8bit computer VGA, USB, and Wifi support).  I'm sure I'm going to need to update the Pico SDK to make that work but I'm not exactly sure what you did originally to bring it into your build system.  If you can provide any insight on that, I would appreciate it.  I'm happy to provide a pull request with the updated code if I get it working either way.

Thanks!

Panda381, 2023-01-31:
I'm sorry, but I'm not entirely sure if I made any changes to the source files as well. I think I just copied the C and S to _sdk, the H header files (and some S) to _sdk\include (preserving partial paths), included the compilation of the C and S source files in Makefile.inc, and included the H header files in global.h.

Panda381, 2023-01-31:
I apologize for not updating a newer version of the SDK - I am creating my own version of the SDK library, with more functionality and more transparency for the programmer. However, it will be completely incompatible with the original SDK library.

codaris, 2023-02-06:
As a test, I tried moving Picovga to the cmake build system that the Pico SDK uses and I've successfully gotten the "Hello World" example to build and run.  I don't know cmake at all but I was able to figure out enough to make it work.

The next test is to try a more complex example and then try and see if I can get the Pico W to work.

codaris, 2023-02-10:
I tested a pretty complex example and everything works except for the one thing I didn't expect to have any problems with: tinyusb host.  It hangs as soon as a device is connected when tuh_task() is called.  Even if I comment out all the code in the tinyusb callbacks it still hangs.  I did not expect that failure.  With the Picovga version of the SDK, and the latest TinyUSB, this same code works perfectly.