Panda381/PicoVGA #18: Do we need vcc pin for vga?

mylab4492, 2023-09-25:
I have used esp32 vga library but vcc pin is not required to be connected.
Also in other library of yours you did not connected the vcc pin but in this library you have connected the vcc pin at pin number 9 with resisitor of 150 ohm.

Panda381, 2023-09-25:
Pin 9 is specified as the power supply for the ROM in the monitor. Sometimes I use it, sometimes I don't, but I haven't noticed a difference. Sometimes I even connect it to GND (easier to connect the connector pins). I think it is not necessary.

mylab4492, 2023-09-25:
Ok, thank you for your quick reply as always.
But somehow I'm not able to get display.
here is what I'm doing
1. compiling vga_baloons program by clicking on c.bat
2.copying program.uf2 to pi pico storage

Note: I'm new to pi pico.

Is there anything wrong here?

Panda381, 2023-09-25:
Check the settings in vga_config.h to see if it matches your hardware - VGA_GPIO_FIRST is the first pin with colors, followed by 8 pins of colors and 1 pin with CSYNC composite sync. However, some monitors do not support composite sync, they require separate HSYNC and VSYNC. This is not included in this version of the library. The PicoVGA description provides a link to a derivative project that includes an adjustment for VSYNC.

Or you can use another (newer) project, PicoLibSDK, which also supports hardware with 8-bit display and a mixture of HSYNC+VSYNC, but uses a simple frame buffer display, and does not support extended modes like PicoVGA. https://github.com/Panda381/PicoLibSDK

mylab4492, 2023-09-25:
Thank you for your suggestions. I'll check this one .

mylab4492, 2023-09-25:
One more question can we rotate the display?

Panda381, 2023-09-25:
PicoVGA cannot rotate the display. It would be difficult to implement - mainly because DMA cannot transfer data with increment other than 1. It would have to be solved by software auxiliary buffer, which would be slow. It would be easier to implement it in drawing routines - they could be modified to render graphics to the frame buffer rotationally. But it wouldn't be easy to modify them.