Panda381/PicoVGA #20: Unable to get display out

KaiyoFox, 2024-11-01:
Im using Codaris's picovga-cmake
And I cannot for the life of me get display out,
I only connected the GND and Hsync/Csync pins
I did previously have the VCC and Color pins hooked up, but I still had no display out
I also tried setting it to TV Mode instead of PC Monitor DEV_VGA

I have my pins set as following:
`#define VGA_GPIO_FIRST	2	// first VGA GPIO
#define VGA_GPIO_NUM	9	// number of VGA GPIOs, including HSYNC and VSYNC
#define VGA_GPIO_OUTNUM	8	// number of VGA color GPIOs, without HSYNC and VSYNC
#define VGA_GPIO_LAST	(VGA_GPIO_FIRST+VGA_GPIO_NUM-1)	// last VGA GPIO
#define VGA_GPIO_SYNC	10	// VGA SYNC GPIO`

In Codaris's project I did discover by default Vsync is enabled, but I commented out the line but still nothing,
I tried wiring it up to use Vsync and still nothing,

I've tried on my newer monitor but I still had little luck

Any idea of what more I could try?

I know it worked at some point, as I was using another pico board, but I fried it or something, I bought a new pico and uploaded the code and now I can't get it to work.
The original pico board I was using was a pico like board from Waveshare

[image]
[image]

KaiyoFox, 2024-11-02:
I think I figured it out,

By mounting my Csync resistor directly to the board, it causes some sorta electromagnetic interference or something with the pico, and prevents the display from understanding the signal, So Solution was to not use the Perfboard, Very odd, I'm not sure what exactly the issue is, but my guess as been stated.

Panda381, 2024-11-02:
Normally, I would advise you that the monitor may not support CSYNC mode only, but requires both HSYNC+VSYNC signals. But as you state, the monitor previously only worked on the CSYNC signal, and it didn't work on the newer monitor either, so that probably shouldn't be the problem. Normally, monitors should only respond to the sync signal, even without the need for a color signal - you should be able to see the black area and see that the monitor is running in a supported video mode. If only the CSYNC signal is used, it is necessary to disable VSYNC support to include vertical sync in the CSYNC signal. To check, it may be a good idea to view the CSYNC signal on an oscilloscope or logic analyzer or in a frequency counter - to see if the frequency matches the specified format. If it does, the monitor should get into sync (even without a video signal). The Pico module itself can also be used to check the signal - just simply count the pulses in 1 second and output the value somewhere, e.g. on a USB console. And yes, reducing the resistance to 47 ohms can also be a good thing to do, some monitors require it.

KaiyoFox, 2024-11-05:
> @KaiyoFox  Which version of pico-sdk do you use? I found out that with newer  pico-sdk than 1.5.1 picovga doesn't work.

Im using the latest so maybe that could also be why
It was working nye even the day prior but I switched to a new pico and it stopped working

but Im thinkingggg the reason is due to soldering it to a board vs prototype which is really odd

closed 2024-11-12 by KaiyoFox